open62541pp/open62541pp #530: add session and node id parameter to methodCallback functions

code-cpp, 2025-01-17:
Recently, we found that in many scenarios, it is necessary to obtain the context information of the method node, such as obtaining the value of the corresponding variable of the object node, so we added session and node ID information to the methodcallback function and tested it.
﻿
About a month ago, we added the NodeID parameter to the ValueCallback and tested it locally without any issues. Since then, the related submissions to improve the interface function of ValueCallback been added, until the recent implementation of the ValueCallbackBase class. I hope to see further updates on the methodcallback related interface.

The following submission is our implementation of the methodcallback function with session and Nodeid information.
```diff
commit 8bf383e8ee11c388029cc887615ac6974e8642e4
Author: zacharyLi <zacharyli@email.cn>
Date:   Fri Jan 17 17:25:35 2025 +0800

    feat: add session and node id parameter to methodCallback functions
    
    Signed-off-by: zacharyLi <zacharyli@email.cn>

diff --git a/examples/events/server_events.cpp b/examples/events/server_events.cpp
index 37dc842..910f89c 100644
--- a/examples/events/server_events.cpp
+++ b/examples/events/server_events.cpp
@@ -14,7 +14,7 @@ int main() {
     objectsNode.addMethod(
         {1, 1000},
         "GenerateEvent",
-        [&](opcua::Span<const opcua::Variant> input, opcua::Span<opcua::Variant>) {
+        [&](opcua::Session&, const opcua::NodeId&, opcua::Span<const opcua::Variant> input, opcua::Span<opcua::Variant>) {
             const auto severity = input.at(0).scalar<uint16_t>();
             const auto& message = input.at(1).scalar<opcua::String>();
 
diff --git a/examples/method/server_method.cpp b/examples/method/server_method.cpp
index a7459cf..88dcdb3 100644
--- a/examples/method/server_method.cpp
+++ b/examples/method/server_method.cpp
@@ -12,7 +12,7 @@ int main() {
     objectsNode.addMethod(
         {1, 1000},
         "Greet",
-        [](opcua::Span<const opcua::Variant> input, opcua::Span<opcua::Variant> output) {
+        [](opcua::Session&, const opcua::NodeId&, opcua::Span<const opcua::Variant> input, opcua::Span<opcua::Variant> output) {
             const auto& name = input.at(0).scalar<opcua::String>();
             const auto greeting = std::string("Hello ").append(name);
             output.at(0) = greeting;
@@ -26,9 +26,9 @@ int main() {
     objectsNode.addMethod(
         {1, 1001},
         "IncInt32ArrayValues",
-        [](opcua::Span<const opcua::Variant> input, opcua::Span<opcua::Variant> output) {
+        [](opcua::Session&, const opcua::NodeId&, opcua::Span<const opcua::Variant> input, opcua::Span<opcua::Variant> output) {
             const auto values = input.at(0).array<int32_t>();
-            const auto delta = input.at(0).scalar<int32_t>();
+            const auto delta = input.at(1).scalar<int32_t>();
             std::vector<int32_t> incremented(values.size());
             std::transform(values.begin(), values.end(), incremented.begin(), [&](auto v) {
                 return v + delta;
@@ -61,5 +61,29 @@ int main() {
         }
     );
 
+    opcua::services::MethodCallback methodCallback =
+    [&](opcua::Session& session, const opcua::NodeId& methodID, opcua::Span<const opcua::Variant>, opcua::Span<opcua::Variant> output) {
+        opcua::Node methodNode(session.connection(), methodID);
+        opcua::Node parentNode = methodNode.browseParent();
+        auto references = parentNode.browseChildren();
+        output.at(0) =  references.size();
+    };
+
+    objectsNode.addMethod(
+        {1, 1002},
+        "ChildNodeNum",
+        methodCallback,
+        { //Input null
+        },
+        {
+            opcua::Argument(
+                "Child Node Num",
+                {"en-US", "Child Node Num"},
+                opcua::DataTypeId::UInt32,
+                opcua::ValueRank::Scalar
+            ),
+        }
+    );
+
     server.run();
 }
diff --git a/include/open62541pp/detail/server_context.hpp b/include/open62541pp/detail/server_context.hpp
index 8987f7e..b470275 100644
--- a/include/open62541pp/detail/server_context.hpp
+++ b/include/open62541pp/detail/server_context.hpp
@@ -17,6 +17,7 @@
 #include "open62541pp/services/detail/monitoreditem_context.hpp"
 #include "open62541pp/types.hpp"  // NodeId, Variant
 #include "open62541pp/ua/types.hpp"  // IntegerId
+#include "open62541pp/services/nodemanagement.hpp"
 
 namespace opcua::detail {
 
@@ -24,7 +25,7 @@ struct NodeContext {
     ValueCallbackBase* valueCallback{nullptr};
     DataSourceBase* dataSource{nullptr};
 #ifdef UA_ENABLE_METHODCALLS
-    std::function<void(Span<const Variant> input, Span<Variant> output)> methodCallback;
+    opcua::services::MethodCallback methodCallback;
 #endif
 };
 
diff --git a/include/open62541pp/services/nodemanagement.hpp b/include/open62541pp/services/nodemanagement.hpp
index 0250920..3c47af3 100644
--- a/include/open62541pp/services/nodemanagement.hpp
+++ b/include/open62541pp/services/nodemanagement.hpp
@@ -18,6 +18,7 @@
 #include "open62541pp/types.hpp"
 #include "open62541pp/ua/nodeids.hpp"  // *TypeId
 #include "open62541pp/ua/types.hpp"
+#include "open62541pp/session.hpp"
 
 namespace opcua {
 class Client;
@@ -570,7 +571,7 @@ auto addPropertyAsync(
  * @param input Input parameters
  * @param output Output parameters
  */
-using MethodCallback = std::function<void(Span<const Variant> input, Span<Variant> output)>;
+using MethodCallback = std::function<void(Session& session, const NodeId& methodID, Span<const Variant> input, Span<Variant> output)>;
 
 /**
  * Add method.
diff --git a/src/services_nodemanagement.cpp b/src/services_nodemanagement.cpp
index 1c964dd..845d8fc 100644
--- a/src/services_nodemanagement.cpp
+++ b/src/services_nodemanagement.cpp
@@ -12,6 +12,14 @@
 
 namespace opcua::services {
 
+static std::optional<Session> getSession(UA_Server* server, const UA_NodeId* sessionId) noexcept {
+    auto* wrapper = asWrapper(server);
+    if (wrapper == nullptr || sessionId == nullptr) {
+        return std::nullopt;
+    }
+    return Session(*wrapper, asWrapper<NodeId>(*sessionId));
+}
+
 AddNodesResponse addNodes(Client& connection, const AddNodesRequest& request) noexcept {
     return UA_Client_Service_addNodes(connection.handle(), request);
 }
@@ -85,10 +93,10 @@ Result<NodeId> addNode<Client>(
 #ifdef UA_ENABLE_METHODCALLS
 
 static UA_StatusCode methodCallback(
-    [[maybe_unused]] UA_Server* server,
-    [[maybe_unused]] const UA_NodeId* sessionId,
+    UA_Server* server,
+    const UA_NodeId* sessionId,
     [[maybe_unused]] void* sessionContext,
-    [[maybe_unused]] const UA_NodeId* methodId,
+    const UA_NodeId* methodId,
     void* methodContext,
     [[maybe_unused]] const UA_NodeId* objectId,
     [[maybe_unused]] void* objectContext,
@@ -97,12 +105,15 @@ static UA_StatusCode methodCallback(
     size_t outputSize,
     UA_Variant* output
 ) noexcept {
-    assert(methodContext != nullptr);
+    assert(methodContext != nullptr && methodId != nullptr);
     const auto* nodeContext = static_cast<opcua::detail::NodeContext*>(methodContext);
     const auto& callback = nodeContext->methodCallback;
-    if (callback) {
+    auto session = getSession(server, sessionId);
+    if (callback != nullptr) {
         return opcua::detail::tryInvoke(
                    callback,
+                   session.value(),
+                   asWrapper<NodeId>(*methodId),
                    Span<const Variant>{asWrapper<Variant>(input), inputSize},
                    Span<Variant>{asWrapper<Variant>(output), outputSize}
         )
diff --git a/tests/services_method.cpp b/tests/services_method.cpp
index 8a87317..19a0421 100644
--- a/tests/services_method.cpp
+++ b/tests/services_method.cpp
@@ -23,7 +23,7 @@ TEST_CASE_TEMPLATE("Method service set", T, Server, Client, Async<Client>) {
         objectsId,
         methodId,
         "Add",
-        [&](Span<const Variant> inputs, Span<Variant> outputs) {
+        [&](opcua::Session&, const opcua::NodeId&, Span<const Variant> inputs, Span<Variant> outputs) {
             if (throwException) {
                 throw BadStatus(UA_STATUSCODE_BADUNEXPECTEDERROR);
             }
```

lukasberbuer, 2025-02-06:
Hi @code-cpp, sorry for the latency.

In my opinion, the current version covers 99% of the use cases and I don't want to complicate everyones code by adding all the available parameters. Both `objectId` and `methodId` can easily be captured with a lambda during creation and accessed within the callback. The session parameters are the only ones, which can not be captured.
Maybe you can describe why capturing doesn't work in your case?

One idea would be, to make the `MethodCallback` a variant with the current simple function signature and the full signature. This wouldn't break any existing code and still enable power users like you to use all possible parameters:

```cpp
using MethodCallback = std::variant<
    std::function<void(Span<const Variant> input, Span<Variant> output)>,
    std::function<void(
        Session& session,
        const NodeId& methodId,
        const NodeId& objectId,
        Span<const Variant> input,
        Span<Variant> output
    )>>;
```

code-cpp, 2025-02-07:
Hi @lukasberbuer 
Thank you very much for your reply. I have a limited understanding of the advanced features of C++,and the easiest solution that comes to mind is to just update the code, ignoring the consistency of our entire project. The use of std::variant that you mentioned is indeed a good solution.

closed 2025-02-07 by code-cpp