open62541pp/open62541pp #537: Assign session context in AccessControl plugin

lukasberbuer, 2025-01-20:
Currently, there is no way to assign a custom session context in the `AccessControl` plugin. Any ideas?

In open62541, a pointer to the context pointer is passed in the `activateSession` callback:
```c
    UA_StatusCode (*activateSession)(UA_Server *server, UA_AccessControl *ac,
                                     const UA_EndpointDescription *endpointDescription,
                                     const UA_ByteString *secureChannelRemoteCertificate,
                                     const UA_NodeId *sessionId,
                                     const UA_ExtensionObject *userIdentityToken,
                                     void **sessionContext);
```
https://github.com/open62541/open62541/blob/v1.3.15/include/open62541/plugin/accesscontrol.h#L43-L48

17steen, 2025-05-12:
AccessControlBase could take in a template parameter with the type of the context? Either that or using std::any for the sessionContext could work i guess. std::any isn't exactly type error prone but at least using the wrong type will have a defined behaviour

lukasberbuer, 2025-05-18:
I like your suggestion using `std::any` 👍

The big challenge seems to be the interoperability with the C API. Users can use the open62541pp wrapper but still create the access control using the C API. Then, the context pointer can be of any type and casting it to `std::any` will be dangerous.

I see two options:
- Keep a registry of registry of session contexts and check for existence before casting it to `std::any` (at least in debug builds)
- Use a sentinel to check if the object was created by the open62541pp access control, e.g.:
  ```cpp
  struct SessionContext {
      static constexpr uint64_t sentinel = 0xC0FFEECAFEBABE42;
      uint64_t id = sentinel;
      std::any context;
  };

  template <typename T>
  T* contextAs(void* context) noexcept {
      auto* typed = static_cast<SessionContext*>(context);
      if (typed->id == SessionContext::sentinel) {
          return std::any_cast<T*>(typed->context);
      } 
      return nullptr;
  }
  ```
  This is not completely safe. If the C context is less then 64 bit, reading the sentinel will overflow the allocated memory. And of course the sentinel is not completely collision free.

What do you think?