open62541pp/open62541pp #538: identifierIf() triggers unreachable code warning with MSVC

pemartin2, 2025-01-21:
Hi @lukasberbuer,

I was playing around with v0.17.0 and just saw that with msvc  version 19.42.34436 
a "warning C4702: unreachable code" is triggered.

Obviously, this is caused by the static_assert together with the last `return nullptr`.

Would that be something you consider to modify/suppress? 

[image]

lukasberbuer, 2025-01-23:
Good catch @pemartin2, thanks!
I just created a PR that should fix the issue #540.

Btw. do you prefer using `NodeId::identifier<T>()` or the `noexcept` version `NodeId::identifierIf<T>`?

closed 2025-01-23 by lukasberbuer

pemartin2, 2025-01-23:
Thank you. Will check the fix as soon as possible.
I definitely see the benefit of `NodeId::identifierIf<T>()`, but for my use case I always know that I have a string identifier. That's why the `NodeId::identifier<T>()` is more handy for me.

Just because you asked for my preferences and use case. What I was actually missing was a wrapper for `UA_NodeId_parse` function to get a NodeId from its string representation, so just the other way around. But not a problem to work-around that.

lukasberbuer, 2025-01-23:
Feedback is highly appreciated. If you have use cases that are not covered by open62541pp, please create an issue or discussion 😊

PR #540 (2025-01-23): fix: unreachable code in NodeId::identifierIf
Closes #538.