open62541pp/open62541pp #539: Question: Callback ownership for ValueCallback

pemartin2, 2025-01-22:
Hi @lukasberbuer 

with #522 you introduced the ValueCallback as abstract base class.

In this PR the code in`server::setVariableNodeValueCallback(const NodeId& id, ValueCallback callback)` changed from
`nodeContext->valueCallback = std::move(callback);`
to
`nodeContext->valueCallback = &callback;`

I.e, the ownership of the callback now stays with the caller.
I just wanted to ask if this is an intended change?

With the old version I could just create my callback, and let it be deleted when going out of scope. With the new version I need to store my callback on the heap. For my use case I actually want is a "create and forget" so I would just need to store the callback for memory reasons.

Just once more, thank you for your great work with this library.

lukasberbuer, 2025-01-22:
Hi @pemartin2,

Yes, with the current implementation, the `ValueCallback` ownership is not transferred to the server.
Other plugins like `AccessControl` have two overloads to specify the ownership:
```cpp
    /// Set custom access control.
    void setAccessControl(AccessControlBase& accessControl);
    /// Set custom access control (transfer ownership to Server).
    void setAccessControl(std::unique_ptr<AccessControlBase>&& accessControl);
```

This could also be done for `ValueCallback`.
What's your take on this? What ownership model do you prefer in your code base?

pemartin2, 2025-01-23:
For the use cases I have  it makes more sense for the library to take ownership. I take the open62541 server as a mock server only.

My use case is quite similar to the one in your example here https://github.com/open62541pp/open62541pp/blob/master/examples/server_valuecallback.cpp. Either I just want to log that a value was changed, i.e. I don't rely on any external data anyways and just want a create and forget. I guess same goes for your example.

The other case I have is to mock, e.g. a Reset. When the reset node receives a `true ` I hold a reference to the server and reset the error code to 0. With the old implementation I just passed the value callback. because the server will outlive the value callback anyways.

Just assuming the other way around, the callback ownership hold externally by the caller. From my understanding I would need to manually ensure that the callback and every reference hold in the callback outlives the server anyways because in other case we will access freed memory.

So from my limited point of view I don't see any use case where I would benefit from keeping the ownership.

lukasberbuer, 2025-01-23:
I agree for the `ValueCallback` plugin, because the logic how to access extern data is embedded in the callbacks.
In my opinion, the story is different for the `DataSource` plugin, because it may encapsulate the data as shown here: https://github.com/open62541pp/open62541pp/blob/master/examples/server_datasource.cpp
I would expect, that users prefer to manage their data (and data sources) outside the server.

Would following overload fulfill your needs?
```cpp
void Server::setVariableNodeValueCallback(const NodeId& id, std::unique_ptr<ValueCallbackBase>&& callback);
```

pemartin2, 2025-01-27:
> Would following overload fulfill your needs?
> 
> void Server::setVariableNodeValueCallback(const NodeId& id, std::unique_ptr<ValueCallbackBase>&& callback);

Yes, definitely.

Not sure about `DataSource` because I was not aware of it and never used it. Not sure what's the difference to `ValueCallback`. From here https://github.com/open62541/open62541/issues/705#issuecomment-233360843 the difference is not quite clear to me.
Wouldn't the lifetime be bound to the lifetime of the server anyways? I mean, the callbacks are called during whole server lifetime potentially... To access the `T data` with another lifetime `T` could always be a `shared_ptr`.
Not 100% sure if I got the intent of `DataSource` right here though...

PR #544 (2025-02-05): feat: move ownership of ValueCallback and DataSource to server
Add overloads to optionally move ownership of the plugins to the server:

```diff
void Server::setVariableNodeValueCallback(const NodeId& id, ValueCallbackBase& callback);
+ void Server::setVariableNodeValueCallback(const NodeId& id, std::unique_ptr<ValueCallbackBase>&& callback);
void Server::setVariableNodeDataSource(const NodeId& id, DataSourceBase& source);
+ void Server::setVariableNodeDataSource(const NodeId& id, std::unique_ptr<DataSourceBase>&& source);
```

Closes #539.
closed 2025-02-05 by lukasberbuer