open62541pp/open62541pp #542: Build error: atomic library issues

felipevillazon, 2025-01-27:
I have been trying to build the library as a pre-compiled library. When trying to run **_cmake --build ._**, I get compilation errors, which are shown below.

[  1%] Building CXX object CMakeFiles/open62541pp.dir/src/client.cpp.o
In file included from /home/felipevillazon/Software/open62541pp/include/open62541pp/detail/open62541/config.h:11,
                 from /home/felipevillazon/Software/open62541pp/build/include_generated/open62541pp/config.hpp:4,
                 from /home/felipevillazon/Software/open62541pp/include/open62541pp/client.hpp:11,
                 from /home/felipevillazon/Software/open62541pp/src/client.cpp:1:
/usr/local/include/open62541/config.h: In function ‘void* UA_atomic_xchg(void**, void*)’:
/usr/local/include/open62541/config.h:237:37: error: expected primary-expression before ‘volatile’
  237 |     return (void *)atomic_exchange((volatile atomic_uintptr_t *)addr,
      |                                     ^~~~~~~~
/usr/local/include/open62541/config.h:237:37: error: expected ‘)’ before ‘volatile’
  237 |     return (void *)atomic_exchange((volatile atomic_uintptr_t *)addr,
      |                                    ~^~~~~~~~
      |                                     )
/usr/local/include/open62541/config.h: In function ‘void* UA_atomic_cmpxchg(void**, void*, void*)’:
/usr/local/include/open62541/config.h:256:37: error: expected primary-expression before ‘volatile’
  256 |     atomic_compare_exchange_strong((volatile atomic_uintptr_t *)addr,
      |                                     ^~~~~~~~
/usr/local/include/open62541/config.h:256:37: error: expected ‘)’ before ‘volatile’
  256 |     atomic_compare_exchange_strong((volatile atomic_uintptr_t *)addr,
      |                                    ~^~~~~~~~
      |                                     )
/usr/local/include/open62541/config.h: In function ‘void* UA_atomic_load(void**)’:
/usr/local/include/open62541/config.h:279:32: error: expected primary-expression before ‘volatile’
  279 |     return (void*)atomic_load((volatile atomic_uintptr_t *)addr);
      |                                ^~~~~~~~
/usr/local/include/open62541/config.h:279:32: error: expected ‘)’ before ‘volatile’
  279 |     return (void*)atomic_load((volatile atomic_uintptr_t *)addr);
      |                               ~^~~~~~~~
      |                                )
gmake[2]: *** [CMakeFiles/open62541pp.dir/build.make:76: CMakeFiles/open62541pp.dir/src/client.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:160: CMakeFiles/open62541pp.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

How can I fix this?

lukasberbuer, 2025-01-27:
These errors are generated by an open62541 header `config.h`. Did you use the bundled open62541 version or did you install open62541 separately? If yes, which version?

Can you please try to compile a simple program just including `open62541/config.h`?

Berendej, 2025-01-28:
Same problem.

In stdatomic.h we have 
```
#if __cplusplus > 202002L
#include <atomic>
using std::atomic_uintptr_t
```

So in oder to compile with config.h
you need to add **-std=c++23**
or 
change generated config.h
as 
```
# elif defined(__has_include) && __has_include(<stdatomic.h>)
#  define UA_HAVE_C11_ATOMICS
#  include <atomic> /* add this */
#  include <stdatomic.h>
using std::atomic_uintptr_t; /* and this */
```

felipevillazon, 2025-01-28:
Hi to both of you, thanks for the fast replies. Before installing open62541pp, I independently installed open62541. Therefore, I decided to uninstall both of them and install open62541pp only. After that, it is working fine.

closed 2025-01-28 by felipevillazon