open62541pp/open62541pp #547: Bug: Server destruction calls UA_Server_run_shutdown  even if server was not started which interfers with enduser Winsockets use

pemartin2, 2025-02-13:
Behavior was introduced with b2727ce9.
reproduced with v0.16.0

### Observed behavior:
The Destruction logic of server interferes with other parts of user program which use Winsockets.

Server is created but run()/runIterate() is never called. As a consequence WSAStartup() from `winsock.h` is never called from within open62541.
Destructor of Server and causes call to `Server::Deleter::operator()` and since b2727ce9 inconditionally calls `UA_Server_run_shutdown` and consequently `WSACleanup `
According to https://learn.microsoft.com/en-us/windows/win32/api/winsock/nf-winsock-wsacleanup

> There must be a call to `WSACleanup `for each successful call to [WSAStartup](https://learn.microsoft.com/en-us/windows/desktop/api/winsock/nf-winsock-wsastartup). Only the final `WSACleanup `function call performs the actual cleanup. The preceding calls simply decrement an internal reference count in the WS2_32.DLL.

By open62541pp calling `WSACleanup` without `WSAStartup` the reference counting becomes wrong and potentially closes sockets which should not be closed.

### Expected behavior
open62541pp number of calls to `WSAStartup` and `WSACleanup`  match.

### Steps to reproduce:
Create Server in an application together with WinSockets.
Call WSAStartup()
Observe how reference counting of sockets goes wrong and Server destructor closes Socket it should not affect.

### Ideas on how to fix
Not sure if we can access `UA_ServerLifecycle` or if this interferes with the threading model used.
The other option would be to make `isRunning` state from context available in the deleter to get the pre b2727ce9 bevhavior.

lukasberbuer, 2025-02-22:
Hi @pemartin2, thank you for the report. This sounds like a serious bug to be fixed.
Ideally I would capture the current misbehavior in a unit test but I have problems to "observe how reference counting of sockets goes wrong". I'm not familiar with WinSockets. Could you please provide a minimal example that is ideally suitable for a test?

pemartin2, 2025-03-10:
@lukasberbuer: Have a look. This one should reproduce the behavior. Hope this helps.
```
#include <doctest/doctest.h>

#include "open62541pp/server.hpp"

#ifdef _WIN32
#include <windows.h>
#include <winsock2.h>
#endif

using namespace opcua;

#ifdef _WIN32
bool isWinsockActive() {
    SOCKET testSocket = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
    if (testSocket == INVALID_SOCKET) {
        int error = WSAGetLastError();
        if (error == WSANOTINITIALISED) {
            return false;  // Winsock is not initialized
        }
    } else {
        closesocket(testSocket);  // Close the test socket if it was created successfully
    }
    return true;  // Winsock is initialized
}

TEST_CASE("Server Winsocket interplay (Windows)") {
    /* Use the MAKEWORD(lowbyte, highbyte) macro declared in Windef.h */
    const WORD wVersionRequested = MAKEWORD(2, 2);
    WSADATA wsaData;
    CHECK(NO_ERROR == WSAStartup(wVersionRequested, &wsaData));
    CHECK(isWinsockActive());

    SUBCASE("Server was run") {
        opcua::Server server{123};
        server.runIterate();  // this calls WSAStartup internally and therefore, it's fine that
                              // ~Server calls WSACleanup(). Reference counting is ok.
    }
    SUBCASE("Server was not run") {
        opcua::Server server{123};
        // server.runIterate();
        // We do not call WSAStartup() but we call WSACleanup() in the destructor
        // WinSock reference counting messed up by the server
    }

    CHECK(isWinsockActive());  // should not be affected by the server
    WSACleanup();
    CHECK(!isWinsockActive());
}
#endif

```

lukasberbuer, 2025-03-10:
That's great! Thanks a lot @pemartin2 👍

PR #556 (2025-03-10): fix: check if server runs before calling UA_Server_run_shutdown
Closes #547
closed 2025-03-10 by lukasberbuer