open62541pp/open62541pp #549: Deallocation failure in opcua::ua::ReadRequest

DmitryYurov, 2025-02-23:
I want to make a service read request to a server in order to read values from several nodes:

```C++
        using opcua::services::detail::createReadRequest;
        using opcua::services::detail::createReadValueId;
        constexpr auto timestamp_type = opcua::ua::TimestampsToReturn::Server;

        opcua::ua::ReadResponse resp;
        {
            auto nodes_to_read = std::array<opcua::ua::ReadValueId, 4>{
                createReadValueId(opcua::NodeId(1, 100115),
                                  opcua::AttributeId::Value),
                createReadValueId(opcua::NodeId(1, 100111),
                                  opcua::AttributeId::Value),
                createReadValueId(opcua::NodeId(1, 100707),
                                  opcua::AttributeId::Value),
                createReadValueId(opcua::NodeId(1, 100719),
                                  opcua::AttributeId::Value),
            };
            // Due to an implementation bug, an attempt to use directly as a parameter causes a premature deallocation
            // const auto req  = createReadRequest(timestamp_type, nodes_to_read);
            const auto rreq = opcua::ua::ReadRequest(createReadRequest(timestamp_type, nodes_to_read));
            resp            = opcua::services::read(client, rreq);
        }
```
In the form presented, the destruction of `rreq` causes the `free` function to be called on invalid pointer:

```
free(): invalid pointer

...

#13
         53:             // const auto req  = createReadRequest(timestamp_type, nodes_to_read);
         54:             const auto rreq = opcua::ua::ReadRequest(createReadRequest(timestamp_type, nodes_to_read));
         55:             resp            = opcua::services::read(client, rreq);
      >  56:         }
         57:
#12   Source "include/open62541pp/ua/types.hpp", line 1419, in ~ReadRequest [0x7fe03f26c93d]
       1416:  * UA_ReadRequest wrapper class.
       1417:  * @see https://reference.opcfoundation.org/Core/Part4/v105/docs/5.10.2
       1418:  */
      >1419: class ReadRequest : public TypeWrapper<UA_ReadRequest, UA_TYPES_READREQUEST> {
       1420: public:
       1421:     using TypeWrapper::TypeWrapper;
#11   Source "include/open62541pp/typewrapper.hpp", line 39, in ~TypeWrapper [0x7fe03f26e383]
         36:         : Wrapper<T>(std::exchange(native, {})) {}
         37: 
         38:     ~TypeWrapper() {
      >  39:         clear();
         40:     };
         41: 
         42:     /// Copy constructor (deep copy).
#10   Source "include/open62541pp/typewrapper.hpp", line 93, in clear [0x7fe03f26ffa5]
         91: protected:
         92:     constexpr void clear() noexcept {
      >  93:         detail::clear(this->native(), UA_TYPES[Index]);
         94:     }
         95: };
#9    Source "include/open62541pp/detail/types_handling.hpp", line 70, in clear<UA_ReadRequest> [0x7fe03f2714c8]
         67:     } else if (isBorrowed(native)) {
         68:         native = {};
         69:     } else {
      >  70:         UA_clear(&native, &type);
         71:     }
         72: }
```

However, uncommenting the line with `createReadRequest` and passing the reference to `ReadRequest` constructor results in a correct deallocation of underlying objects:

```C++
            const auto req  = createReadRequest(timestamp_type, nodes_to_read);
            const auto rreq = opcua::ua::ReadRequest(req);
            resp            = opcua::services::read(client, rreq);
```

Version info:
open625141pp version: 0.17.0
open625141 version: 1.3.15
compiler version: gcc 11.4.0
OS version: Ubuntu 22.04.3 LTS

lukasberbuer, 2025-02-24:
Hi @DmitryYurov,
You are using functions from the `detail` namespace.

> A namespace called `detail` is typically used in just the same way as a namespace called `internal`. It's for "internal details" that are not visible in the public interface or which, at least, should be ignored by external users.
> https://stackoverflow.com/a/26546780/9967707

The `opcua::services::detail::createReadValueId` is optimized to avoid heap allocations and will reference the passed `NodeId`. In your case, the `NodeId` is passed as a rvalue which will go out of scope after the function call. Please don't use that function. Also, the function may change any time because it's an internal detail.

Why don't you just use the `ReadValueId` constructor (and skip the array template args with CTAD 😊)?
```cpp
auto nodes_to_read = std::array{
    opcua::ua::ReadValueId{opcua::NodeId{1, 100115}, opcua::AttributeId::Value},
    opcua::ua::ReadValueId{opcua::NodeId{1, 100111}, opcua::AttributeId::Value},
    opcua::ua::ReadValueId{opcua::NodeId{1, 100707}, opcua::AttributeId::Value},
    opcua::ua::ReadValueId{opcua::NodeId{1, 100719}, opcua::AttributeId::Value},
};
```

DmitryYurov, 2025-02-25:
Hi @lukasberbuer,

Thank you for your response.
I tried your suggestion, but unfortunately the problem still persists.

It is reliably reproducible with

- passing rvalue reference resulting from `opcua::services::detail::createReadRequest` directly into `opcua::ReadRequest` constructor without prior storing it in a variable and later calling `opcua::services::read(client, rreq)`, or
- calling a service read function without creating intermediate read request, e.g. `opcua::services::read(client, nodes_to_read, timestamp_type)`.

lukasberbuer, 2025-02-25:
Great catch @DmitryYurov, thank you!
I could catch the deallocation failure with ASAN in the added test in #552.

PR #552 (2025-02-25): fix: deallocation failure in read/write requests
Closes #549.
closed 2025-02-25 by lukasberbuer