open62541pp/open62541pp #555: New types of async callback methods

OgreTransporter, 2025-03-02:
The callback types have been changed in the current version 1.4.10. See:
https://github.com/open62541/open62541/commit/de6e63bfffc5972f83be70597ed5c07ca0fcebf7

Here is an example proposal for the deleteSubscriptionsAsync function:
```
/**
 * @copydoc deleteSubscriptions
 * @param token @completiontoken{void(DeleteSubscriptionsResponse&)}
 * @return @asyncresult{DeleteSubscriptionsResponse}
 */
template <typename CompletionToken>
auto deleteSubscriptionsAsync(
    Client& connection, const DeleteSubscriptionsRequest& request, CompletionToken&& token
) {
    return detail::AsyncServiceAdapter<DeleteSubscriptionsResponse>::initiate(
        connection,
#if (UA_OPEN62541_VER_MAJOR == 1 && UA_OPEN62541_VER_MINOR == 4 && UA_OPEN62541_VER_PATCH >= 9) || (UA_OPEN62541_VER_MAJOR == 1 && UA_OPEN62541_VER_MINOR > 4) || UA_OPEN62541_VER_MAJOR > 1
        [&](UA_ClientAsyncDeleteSubscriptionsCallback callback, void* userdata) {
#else
        [&](UA_ClientAsyncServiceCallback callback, void* userdata) {
#endif
            throwIfBad(UA_Client_Subscriptions_delete_async(
                opcua::detail::getHandle(connection), asNative(request), callback, userdata, nullptr
            ));
        },
        std::forward<CompletionToken>(token)
    );
}
```

OgreTransporter, 2025-03-02:
I've already implemented the changes in my fork, but it's not enough for a PR because more parts, such as the tests, have to be adapted.

https://github.com/OgreTransporter/open62541pp/commit/dd734e526af787df84e7e64abaa9d2e7d5a52aa3

lukasberbuer, 2025-03-25:
Thank you @OgreTransporter!
The typed async callbacks seem to be only in master at the moment and not in v1.4.10 and v1.4.11:
https://github.com/open62541/open62541/blob/v1.4.10/include/open62541/client_subscriptions.h
I would like to wait until they are finally released in a new v1.4 version to also test this in CI.