open62541pp/open62541pp #557: Q:How subscription and read/write use the same client

PandaBoby, 2025-03-17:
> When I use subscription the current thread gets blocked, how can I make a client subscribe, read and write at the same time

`
void connect()
    {
        while (RUNNING)
        {
            try
            {
                m_client_.connect(m_endpoint_);
                break;
            }
            catch (const std::exception& e)
            {
                m_client_.disconnect();
                LOG_E("opcua connection {} error: {}, retry after {} seconds", m_endpoint_, e.what(), 5);
                std::this_thread::sleep_for(std::chrono::seconds(5));
            }
        }
    }`

 ====================

    void setSubscription(CUMap<Str, Str>& nodes)
    {

        m_client_.onSessionActivated([&]
        {

            m_subscription_ = std::make_unique<opcua::Subscription<opcua::Client>>(m_client_);

            Vector<opcua::MonitoredItem<opcua::Client>> monitored_items_;
            opcua::SubscriptionParameters subscriptionParameters{};
            subscriptionParameters.publishingInterval = 200;
            subscriptionParameters.maxKeepAliveCount = 1000;
            m_subscription_->setSubscriptionParameters(subscriptionParameters);
            m_subscription_->setPublishingMode(true);

            for (const auto& [k, v] : nodes)
            {
                monitored_items_.push_back(subscribeDataChangeHandler(k, v));
            }

            opcua::MonitoringParametersEx monitoringParameters{};
            monitoringParameters.samplingInterval = 100.0;

            for (auto& item : monitored_items_)
            {
                item.setMonitoringParameters(monitoringParameters);
                item.setMonitoringMode(opcua::MonitoringMode::Reporting);
            }

            m_client_.run();
        });

        connect();
    }

`void setInt16(opcua::NamespaceIndex index, CStrView& node, CInt16& value)
    {
        try
        {
            reconnect();
            opcua::Node nodeId(m_client_, {index, node});
            const auto dt = nodeId.writeValueScalar<Int16>(value);
        }
        catch (const std::exception& e)
        {
            LOG_E("setInt16 error: {}", e.what());
        }
    }`

lukasberbuer, 2025-04-01:
open62541 uses a single event loop to process all operations. `Client::run()` will start the event loop and block the current thread. If you want to process the subscriptions but read/write at the same time, there are two simple options:
1. Use a second thread AND use open62541 v1.4 for a thread-safe client API
2. Interleave operations by running them in between `Client::runIterate()` calls

PandaBoby, 2025-04-08:
I'm sorry, I've checked the information, but I'm still not sure how I should implement it, could you provide me with a use case.
I simply need to batch subscribe to the data and be able to do random reads and random writes int16 is sufficient

lukasberbuer, 2025-05-01:
This example shows how to run the client in another thread so you can to reads/writes from the main thread:
https://github.com/open62541pp/open62541pp/blob/master/examples/method/client_method_async.cpp

closed 2025-05-01 by lukasberbuer