open62541pp/open62541pp #598: opcua::services::MethodCallback should contain the same arguments as UA_MethodCallback

17steen, 2025-04-29:
If specific access control has to be done before executing a method, as i currently understand, there is no way to get a handle to the user session from within the method.

e.g. user should not be able to call a method with specific arguments

17steen, 2025-04-29:
This came up because it seems that setting the UserExecutable flag to false doesn't actually restrict the use of a method by a client (testing with UAExpert), and there should therefore be an extra check within the method callback that the user could call it.

lukasberbuer, 2025-04-29:
I understand... `UA_MethodCallback` is defined as:

```c
typedef UA_StatusCode
(*UA_MethodCallback)(UA_Server *server, const UA_NodeId *sessionId,
                     void *sessionContext, const UA_NodeId *methodId,
                     void *methodContext, const UA_NodeId *objectId,
                     void *objectContext, size_t inputSize,
                     const UA_Variant *input, size_t outputSize,
                     UA_Variant *output);
```

Currently open62541pp only uses:
- `size_t inputSize`, `const UA_Variant* input` -> `opcua::Span<const opcua::Variant>`
- `size_t outputSize`, `UA_Variant* output` -> `opcua::Span<opcua::Variant>`

We could extend the signature to also capture:
- `UA_Server* server`, `const UA_NodeId* sessionId`, `void* sessionContext` -> `opcua::Session`
- if required: `const opcua::NodeId& methodId`, `const opcua::NodeId& objectId`

The `methodContext` is used internally and shouldn't be exposed to the library user.
Is the added `opcua::Session` parameter enough for your use case or should we also integrate `methodId` and `objectId`?

To be backwards compatible and allow simple signatures, I would propose to use a `std::variant`, e.g.:

```cpp
using MethodCallback = std::variant<
    std::function<void(Span<const Variant> input, Span<Variant> output)>,
    std::function<void(Session& session, Span<const Variant> input, Span<Variant> output)>
>;
```

lukasberbuer, 2025-04-29:
Already discussed in #530.

17steen, 2025-04-29:
For my current use case, only the session would be necessary, but i think allowing all members to be exposed wouldn't be a bad thing. I think the variant alternative in the previously discussed issue would work.

PR #601 (2025-05-01): feat: MethodCallback with session, methodId and objectId
Support full function signature for method callbacks with `session`, `methodId` and `objectId`.
The type `MethodCallback` is now a variant that takes either the simple signature with only input and output arguments or the full signature.

Full signature:
```cpp
StatusCode(
    Session& session,
    Span<const Variant> input,
    Span<Variant> output,
    const NodeId& methodId,
    const NodeId& objectId
)>
```

Closes #598.
closed 2025-05-01 by lukasberbuer