open62541pp/open62541pp #609: Double lock when reading a sessionAttribute in the `allowBrowseNode` method of an access control class.

17steen, 2025-05-08:
If an access control object queries a session attribute in the allowBrowseNode method, the program will SIGABRT due to an invalid assertion:

UA_LOCK: Assertion `++(lock->mutexCounter) == 1' failed.

I have written a minimal reproducible example:

```cpp
    auto port = uint16_t{4840};

    struct my_custom_access_control final : opcua::AccessControlDefault
    {
        my_custom_access_control() : AccessControlDefault(true, {})
        {
        }

        opcua::StatusCode activateSession(opcua::Session& session, const opcua::EndpointDescription& endpointDescription,
                                          const opcua::ByteString& secureChannelRemoteCertificate, const opcua::ExtensionObject& userIdentityToken) override
        {
            session.setSessionAttribute({0, "my_custom_attribute"}, opcua::Variant{"anonymous"});
            return AccessControlDefault::activateSession(session, endpointDescription, secureChannelRemoteCertificate, userIdentityToken);
        }

        bool allowBrowseNode(opcua::Session& session, [[maybe_unused]] const opcua::NodeId& nodeId) override
        {
            auto attribute = session.getSessionAttribute({0, "my_custom_attribute"}).to<std::string>();
            return attribute == "anonymous";
        }
    };

    // auto access_control = integration::opcua_access_control_t{};
    auto access_control = my_custom_access_control{};

    // make a server with default access control
    auto server_config = opcua::ServerConfig{port};
    server_config.setAccessControl(access_control);
    auto server = opcua::Server{std::move(server_config)};

    auto objects_node = opcua::Node{server, opcua::ObjectId::ObjectsFolder};

    // hello world string
    auto variable_attributes = opcua::VariableAttributes{};
    variable_attributes.setValueScalar("Hello world!");
    objects_node.addVariable({1, "hello_world"}, "hello_world", variable_attributes);

    // run server in another thread
    auto server_thread = std::jthread{[&server]() {
        server.run();
    }};
    while (!server.isRunning())
    {
        std::this_thread::yield();
    }

    {
        auto client_config = opcua::ClientConfig{};
        auto client = opcua::Client{std::move(client_config)};
        client.connect(fmt::format("opc.tcp://localhost:{}", port));

        auto client_object_node = opcua::Node{client, opcua::ObjectId::ObjectsFolder};

        for (auto&& node: client_object_node.browseChildren())
        {
            std::cerr << "Node: " << opcua::toString(node.id()) << std::endl << "Name: " << node.readBrowseName().name() << std::endl;
        }
    }

    server.stop();
```

17steen, 2025-05-09:
I recreated the same example locally with the c library (1.3.15). and it also gets this issue, so it seems to be an issue coming from the c library.

Note: I noticed that with 1.3.15, getSessionParameter is called, instead of UA_Server_getSessionAttribute in 1.4.11, this might be the issue? not sure
```c
    UA_Server_getSessionParameter(server, sessionId, "my_custom_attribute", my_custom_attribute_variant.handle());
    auto my_custom_attribute = my_custom_attribute_variant.to<std::string>();
```

lukasberbuer, 2025-05-09:
Thanks for diving deep again @17steen. Would you open again an open62541 issue?
The naming shouldn't be a problem. The session attributes of v1.4 were previously named session parameters in v1.3.

Are you using session attributes to provide additional data to the access control callbacks?
open62541 has another option by assigning a session context in the `activateSession` callback as described in #537. I didn't figure out a good type-safe and modern way to port this to C++17. Maybe you have an idea here.

17steen, 2025-05-09:
Yes, I want some nodes to not be displayed if a user isn't logged in with certain credential, and I believe this is done by restricting browsability, i will look into the session context.

17steen, 2025-05-09:
i can't submit my c example because it looks like this: 
```cpp
UA_StatusCode activateSession(UA_Server* server, UA_AccessControl* ac, const UA_EndpointDescription* endpointDescription,
                              const UA_ByteString* secureChannelRemoteCertificate, const UA_NodeId* sessionId, const UA_ExtensionObject* userIdentityToken,
                              void** sessionContext)
{
    std::ignore = ac;
    auto serverWrapper = opcua::Server{server};
    auto sessionWrapper = opcua::Session{serverWrapper, opcua::NodeId{*sessionId}, sessionContext};
    auto endpointDescriptionWrapper = opcua::asWrapper<opcua::EndpointDescription>(endpointDescription);
    auto secureChannelRemoteCertificateWrapper = opcua::asWrapper<opcua::ByteString>(secureChannelRemoteCertificate);
    auto userIdentityTokenWrapper = opcua::asWrapper<opcua::ExtensionObject>(userIdentityToken);

    sessionWrapper.setSessionAttribute({0, "my_custom_attribute"}, opcua::Variant{"anonymous"});

    return opcua::AccessControlDefault{}.activateSession(sessionWrapper, *endpointDescriptionWrapper, *secureChannelRemoteCertificateWrapper,
                                                         *userIdentityTokenWrapper);
}

static UA_Boolean allowBrowseNode(UA_Server* server, UA_AccessControl* ac, const UA_NodeId* sessionId, void* sessionContext, const UA_NodeId* nodeId,
                                  void* nodeContext)
{

    auto my_custom_attribute_variant = opcua::Variant{};
    UA_Server_getSessionParameter(server, sessionId, "my_custom_attribute", my_custom_attribute_variant.handle());
    auto my_custom_attribute = my_custom_attribute_variant.to<std::string>();

    std::ignore = ac;
    std::ignore = nodeId;
    std::ignore = nodeContext;
    std::ignore = sessionContext;
    return true;
}

void setCustomAccessControl(UA_ServerConfig* config)
{
    /* Use the default AccessControl plugin as the starting point */
    UA_Boolean allowAnonymous = true;
    UA_String encryptionPolicy = config->securityPolicies[config->securityPoliciesSize - 1].policyUri;
    config->accessControl.clear(&config->accessControl);
    UA_AccessControl_default(config, allowAnonymous, nullptr, &encryptionPolicy, 0, NULL);

    /* Override accessControl functions for nodeManagement */
    // config->accessControl.activateSession = activateSession;
    config->accessControl.allowBrowseNode = allowBrowseNode;
}

```

I still used a little bit of the c library

lukasberbuer, 2025-05-15:
You can use `UA_AccessControl_default` and later change the `allowBrowseNode` context:

```c
#include <open62541/plugin/accesscontrol_default.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

static UA_Boolean allowBrowseNode(
    UA_Server* server,
    [[maybe_unused]] UA_AccessControl* ac,
    const UA_NodeId* sessionId,
    [[maybe_unused]] void* sessionContext,
    [[maybe_unused]] const UA_NodeId* nodeId,
    [[maybe_unused]] void* nodeContext
) {
    UA_Variant attribute;
    UA_Variant_init(&attribute);
    UA_Server_getSessionParameter(server, sessionId, "my_custom_attribute", &attribute);
    return true;
}

static void setCustomAccessControl(UA_ServerConfig* config) {
    UA_Boolean allowAnonymous = true;
    UA_AccessControl_default(config, allowAnonymous, nullptr, nullptr, 0, nullptr);
    config->accessControl.allowBrowseNode = allowBrowseNode;
}

int main() {
    UA_ServerConfig config;
    UA_ServerConfig_setDefault(&config);
    setCustomAccessControl(&config);
    UA_Server* server = UA_Server_newWithConfig(&config);

    volatile bool running = true;
    UA_Server_run(server, &running);
}
```

17steen, 2025-05-16:
https://github.com/open62541/open62541/issues/7260 created an issue here