open62541pp/open62541pp #610: examples/access_control/server_access_control.c: Doesn't work when using latest open62541 release (1.4.11)

17steen, 2025-05-09:
Not sure how relevant since 1.4.11 is not the default version used for this library.
But I wanted to use it and see if some other issues were gone.

steps to reproduce: 

1. checkout 1.4.11 version of open62541, run `server_accesscontrol` example.
2. try to log in with an opc ua client (e.g. uaexpert)
3. try user/user, anonymous, admin/admin

Gets this error: 
```
Error 'BadIdentityTokenInvalid' was returned during ActivateSession
```

None of the logins  work, anonymous login also doesn't work.

Tested the C library example and i can login with peter/peter123.

lukasberbuer, 2025-05-09:
Yes this is weird. Version 1.4.6 is tested in CI and works. Somewhere between 1.4.6 and 1.4.11, a patch release broke this. I have to dive deeper...

17steen, 2025-05-12:
I pulled master, went into the submodule and checked out 1.4.6, and I also get the same issue trying to connect with UAExpert, am i doing something wrong? 🤔

17steen, 2025-05-12:
```
Connection status of server 'open62541-based OPC UA Application@localhost' changed to 'Disconnected'.
Endpoint: 'opc.tcp://localhost:4840'
Security policy: 'http://opcfoundation.org/UA/SecurityPolicy#None'
ApplicationUri: 'urn:open62541.server.application'
Used UserTokenType: Anonymous
The server returned no certificate, all certificate checks will be skipped.
Error 'BadIdentityTokenInvalid' was returned during ActivateSession
Connection status of server 'open62541-based OPC UA Application@localhost' changed to 'Disconnected'.
```

17steen, 2025-05-12:
> Yes this is weird. Version 1.4.6 is tested in CI and works. Somewhere between 1.4.6 and 1.4.11, a patch release broke this. I have to dive deeper...

I think it must be due to this commit: https://github.com/open62541/open62541/commit/be89dbdbc169fb4590041bbc6e6f5d7a2201c9e2

17steen, 2025-05-12:
https://github.com/open62541/open62541/blob/be8449104cfaabb466041ca6bea8e9d8b7d35404/src/server/ua_services_session.c#L550 This check here fails when using the .cpp example, but passes when using the c example.

17steen, 2025-05-12:
[image] 
utPolPostFix contains `open62541-anonymous-policy` when run from the cpp example, but it seems that it should contain `#None`.

17steen, 2025-05-12:
[image]
One difference i can spot is the following, while the default access control in the c library sets a default security policy (`http://opcfoundation.org/UA/SecurityPolicy#None`), the wrapper library leaves it empty.

17steen, 2025-05-12:
https://github.com/open62541/open62541/blob/be8449104cfaabb466041ca6bea8e9d8b7d35404/src/server/ua_services_discovery.c#L388 at this line, the `#None` postfix is appended, this must somehow not happen when the wrapper is in use.

17steen, 2025-05-12:
> If i remove the call to `copyUserTokenPoliciesToEndPoints` in `ServerConfig::setAccessControl`, the `AccessControlCustom::activateSession` method actually gets called.
> 
> But setting a session attribute fails, because the inner session attribute map is never created? (it's null)

https://github.com/open62541/open62541/pull/6970/commits It seems like this was fixed before but never made it into a release? It seems as though we can only rely on the session context so far.

lukasberbuer, 2025-05-15:
Thank you @17steen for your deep dive. It seems to be a combination of following issues:
1. Added `#None` suffix to policy id as mentioned by you. Fixed with https://github.com/open62541pp/open62541pp/pull/612/commits/738e4e49c717328f77c73d8e4ad6eb33ac95379e
2. Weird issue when user token policies are copied to the endpoints in v1.4. Copying the user token policies isn't required anyway. The defined user token policies in access control are automatically looked up if the user token policy array in the endpoints is empty. Fixed with https://github.com/open62541pp/open62541pp/pull/612/commits/ce5bb8a3890b461a00b96d4ff5448c8cc8a9946e
3. v1.4 introduced a new server setting `allowNonePolicyPassword` which is by default `false`. You need to allow the transfer of login credentials without encryption explicitly by setting this setting to `true`. Fixed in https://github.com/open62541pp/open62541pp/pull/612/commits/5b7ca369a66a9ebd10b30b6f4e85bfe79031062a

Can you please test/review the PR #612? The session attribute issue sadly remains until fixed in open62541.

closed 2025-05-15 by lukasberbuer

17steen, 2025-05-15:
Thanks for fixing these issues!

about the session attributes: i created #611 which is implemented in a pretty ugly way but works as a proof of concept, i have tested it and it works

PR #612 (2025-05-17): fix: access control with v1.4
Fixes two issues:
- `Error 'BadIdentityTokenInvalid' was returned during ActivateSession` in `server_accesscontrol` example (closes #610)
- `BadIdentityTokenInvalid` in CI with open62541 >= v1.4.6