open62541pp/open62541pp #627: client::run() no longuer works on open62541 v1.4.12

17steen, 2025-05-23:
If one runs: examples_server together with examples_client_subscription using open62541 v1.4.12., the client will fail to run() due to this error: 

`error/eventloop	Cannot run EventLoop from the run method itself.`

Since subscriptions and async functionality requires the client to run, they no longuer work.

Sample log from `examples_client_subscription`: 
```
[2025-05-23 11:59:35.902 (UTC+0200)] info/network	TCP 4	| Opening a connection to "localhost" on port 4840
[2025-05-23 11:59:35.903 (UTC+0200)] info/channel	TCP 4	| SC 4	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2025-05-23 11:59:35.903 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-05-23 11:59:35.903 (UTC+0200)] warn/client	The server returned Endpoints with a different EndpointUrl opc.tcp://qcify-robin-laptop:4840 than was used to initialize the connection: opc.tcp://localhost:4840. Some servers require a complete match of the EndpointUrl/DiscoveryUrl (including the path) to return all endpoints.
[2025-05-23 11:59:35.903 (UTC+0200)] info/client	Selected endpoint with EndpointUrl opc.tcp://localhost:4840, SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2025-05-23 11:59:35.905 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2025-05-23 11:59:35.905 (UTC+0200)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2025-05-23 11:59:35.905 (UTC+0200)] error/eventloop	Cannot run EventLoop from the run method itself
[2025-05-23 11:59:35.905 (UTC+0200)] warn/client	Request with unknown RequestId 27
[2025-05-23 11:59:35.905 (UTC+0200)] error/client	Processing the message returned the error code BadSecurityChecksFailed
[2025-05-23 11:59:35.905 (UTC+0200)] info/channel	TCP 4	| SC 4	| SecureChannel closed
[2025-05-23 11:59:35.905 (UTC+0200)] info/client	Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2025-05-23 11:59:35.905 (UTC+0200)] info/network	TCP 4	| Socket closed```

lukasberbuer, 2025-05-23:
Hi @17steen,
You get the same error if you run the original open62541 examples [client_subscription_loop](https://github.com/open62541/open62541/blob/v1.4.12/examples/client_subscription_loop.c) with [server_mainloop](https://github.com/open62541/open62541/blob/v1.4.12/examples/server_mainloop.c).
Hope this gets fixed soon in open62541.

PandaBoby, 2025-05-27:
Try using the following code:

`subscribe();
while (true)
        {
            try
            {
                m_client.connect(OPCUA_SERVER_ENDPOINT);
                m_client.runIterate(100);
                break;
            }
            catch (CException& e)
            {
                LOG_E("err:{}", OPCUA_SERVER_ENDPOINT, e.what());
                std::this_thread::sleep_for(std::chrono::seconds(5));
            }
        }`

ErikTempelaarVO, 2025-06-30:
@17steen 
Thanks for raising this issue here and making me aware of it.
I wanted to upgrade from 1.3.15 because of other issues, but now I'm running into this one.

Is there an upstream bug I could follow, or any known work arounds?
I tried with 1.4.6 and 1.4.11.1 and the output is similar.

ErikTempelaarVO, 2025-06-30:
Here's an (unmerged) fix for the open62541 example code:
https://github.com/open62541/open62541/issues/5816

"A short explanation.
The EventLoop is the responsible for the control flow.
Callbacks from inside the EventLoop can only use the async API of open62541.
Otherwise we have blocking calls calling other blocking calls.
And that gets ugly fast."

So it seems you cannot create subscriptions from inside callbacks, unless you use the async-api from open62541?

closed 2025-11-23 by lukasberbuer