open62541pp/open62541pp #660: Certificate authentication

davidhjp01, 2025-08-07:
Hi, I am trying to authenticate using certificate and private key with server implemented using Eclipse Milo.

I have configured the client as follow
```cpp
opcua::ClientConfig config(certificate, privateKey, {}, {});
config.setSecurityMode(opcua::MessageSecurityMode::SignAndEncrypt);
config.setUserIdentityToken(opcua::X509IdentityToken{certificate});
UA_CertificateVerification_AcceptAll(&config.handle()->certificateVerification);
```
Private key is in `.pem` and tried certificate with both `pem` and `der` format, but neither works and receiving `BadIdentityTokenInvalid`:
```
[2025-08-07 15:26:58.462][info][client] Selected endpoint 0 in URL opc.tcp://localhost:4840/ with SecurityMode SignAndEncrypt and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2025-08-07 15:26:58.469][info][client] Selected UserTokenPolicy certificate with UserTokenType Certificate and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2025-08-07 15:26:58.477][info][client] Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2025-08-07 15:26:58.542][info][securechannel] Connection 780 | SecureChannel 741499321 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 and a revised lifetime of 600.00s
[2025-08-07 15:26:58.551][info][client] Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2025-08-07 15:26:59.197][info][client] Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2025-08-07 15:26:59.211][info][client] Received a ServiceFault response
[2025-08-07 15:26:59.216][info][client] The ServiceResult has the StatusCode BadIdentityTokenInvalid
[2025-08-07 15:26:59.223][error][client] ActivateSession failed with error code BadIdentityTokenInvalid
[2025-08-07 15:26:59.230][info][client] Client Status: ChannelState: Open, SessionState: ActivateRequested, ConnectStatus: BadIdentityTokenInvalid
```

The server already have the client certificate in the trust list.

Any idea what could be an issue?

RudolfRendier, 2025-08-07:
Looks like this requires open62541 v1.4:
https://github.com/open62541/open62541/issues/4977

Onwcan, 2026-09-09:
@davidhjp01 

Hi,

`ClientConfig::setAuthenticationCertificate` was just merged in #729, which should address this.

The cause was that open62541 needs the **private key** to create the `userTokenSignature` that proves you own the certificate. `setUserIdentityToken(X509IdentityToken{certificate})` only sets the token — there was no way to supply the corresponding key through the C++ API, so the server had nothing to verify and rejected the session with `BadIdentityTokenInvalid`.

On current master you can write:

```cpp
opcua::ClientConfig config(certificate, privateKey, {}, {});
config.setSecurityMode(opcua::MessageSecurityMode::SignAndEncrypt);
config.setAuthenticationCertificate(userCertificate, userPrivateKey);